DC-analysis/DCscope #7: Cut axis labels when exporting subplot as PNG

chrherold, 2019-12-03:
Axis label are cut when a subplot is exported as PNG file. 

Example below, red Arrows indicate a cut of part of the panel to the left on the left side and the cut off label of this plot on the right side. SVG export works correctly though.

[image]

chrherold, 2019-12-03:
Additional remark: Ideally, every exported subplot should also present the names of x and y axis. Further use would otherwise be limited.

paulmueller, 2019-12-03:
I guess it would make sense to then create a completely new plot in the background with
- [ ] title
- [ ] axes labels
- [ ] same (or at least similar) size / aspect ratio

chrherold, 2019-12-03:
Seems very reasonable, yes. At least in case of subplot export.

paulmueller, 2020-01-16:
Please check whether this is resolved in https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/releases/tag/2.0.0b12

paulmueller, 2020-01-16:
But it should be in b13

closed 2020-01-16 by paulmueller