DC-analysis/DCscope #11: Compute Statistics - which filter settings to use

chrherold, 2019-12-04:
'all data located in a directory':
This is an issue to discuss which filter settings should be applied when statistics are to be computed on 'all data located in a directory' as can be chosen in the dialog field of the 'Compute'/'Statistics' feature.

I would suggest to apply all filter that are checked as 'use this filter' in an opened session.
Any Feedback would be nice.

'all datasets of the current session' :
If Statistics are computed for 'all datasets of the current session' the computation should be carried out for all individually activated filters for a specific dataset

AND

only for those datasets that are checked as 'use this dataset' (this might currently not be the case)

paulmueller, 2019-12-04:
I think it would be better to let the user choose which filter ray (yes I just made that term up) to use (e.g. that of dataset 1 or so). This way, the user does not have to change the pipeline to apply the same filters of a dataset within a session to other datasets. Does that sound acceptable?

chrherold, 2019-12-04:
Yes it does for me. 👍

commit f1e13e47 (2019-12-04): feat: allow to select filter ray when computing statistics (close #11)
closed 2019-12-04 by paulmueller