DC-analysis/DCscope #18: Switch of selected Filter Set in Analysis view after 'apply' is executed

chrherold, 2019-12-04:
With multiple filter sets available, the first filter in the drop down menu is automatically selected once apply is executed in another filter set. This is very irritating.

paulmueller, 2019-12-04:
Yeah, it should also jump to the current QuickView selection btw. But I also find that behavior irritating at times.

chrherold, 2019-12-04:
Ahh, this makes sense, I had the first filter set active in quick view.

And the fact it switches when quick view is changed to a different field is very much appreciated in the general work flow. 

Maybe it should just not jump there if quick view selection is kept steady but something is changed in a 'quickview non-selected' filter set.

paulmueller, 2019-12-04:
ok

commit 81370b70 (2019-12-04): fix: do not change filter selection automatically when hitting apply (close #18)
closed 2019-12-04 by paulmueller