DC-analysis/DCscope #21: Remove Plot - OSX - Error Message

chrherold, 2019-12-04:
The following error message is shown when removing a plot in the osx version:

Unhandled exception in Shape-Out version 2.0.0a27:
Traceback (most recent call last):
  File "shapeout2/gui/pipeline_plot.py", line 32, in update_content
  File "shapeout2/pipeline/core.py", line 403, in get_plot_datasets
KeyError: 'Plot_2'

However, the plot is removed successfully from the Block Matrix and the plot window is closed
as expected.

paulmueller, 2019-12-04:
This is another iteration on the original problem you described. I believe I have fixed it in 2.0.0b2.

FYI: 2.0.0b1 actually introduced another problem with removing plots.

Please close this issue if the problem is resolved.

chrherold, 2019-12-06:
Fixed in 2.0.0b2.

closed 2019-12-06 by chrherold