DC-analysis/DCscope #28: OSX inferface specific - update view of Block Matrix after first Plot added

chrherold, 2019-12-11:
OSX version:

When a plot is added for the first time in a session, the plot column in the Block Matrix does not readily appear. Selecting a different program or rescaling the Block Matrix window leads to correct rebuilding of the Block Matrix. 

Behavior for further plots added is as expected.

paulmueller, 2019-12-11:
I tried to fix it in 58c9cce. Please check again in 2.0.0b5.

chrherold, 2019-12-11:
Unfortunately, still the same behavior.

paulmueller, 2019-12-11:
Please check again in 2.0.0b6.

paulmueller, 2019-12-16:
Please check again in 2.0.0b10.

chrherold, 2019-12-17:
No change in b10 either.

paulmueller, 2020-01-16:
I cannot reproduce this on my macOS test machine. However. I run macOS 10.11.6 and are forced to use PyQt5 5.9...
Is this still an issue for you with b11?

chrherold, 2020-01-16:
Seems to have disappeared with b11! Nice, do you know why?

paulmueller, 2020-01-16:
No

commit 6481f799 (2019-12-11): try again to fix #28
closed 2020-01-16 by paulmueller