DC-analysis/DCscope #29: OSX interface - disable independent 'close window' buttons in header of windows

chrherold, 2019-12-11:
In OSX, the 'close' button in the header of windows like Plot, Analysis View and Quick View is active and can be used to close a window (red button). 

[image]

This leads to loss of content when the window is opened again. Windows are only gray.

[image]

I would suggest disabling the 'close' button.

paulmueller, 2019-12-11:
I tried to fix it in f169ab5. Let me know if this still persists in 2.0.0b5.

chrherold, 2019-12-12:
Close buttons are still alive and kicking in 2.0.0b7. :/

paulmueller, 2019-12-13:
What about 2.0.0b7? Could you please check all windows (QuickView, AnalysisView and plots)?

chrherold, 2019-12-13:
same issue ... for all three window types

paulmueller, 2019-12-13:
b? https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/releases/tag/2.0.0b9

paulmueller, 2019-12-13:
Note that in b9, not all windows do have the title set. I used different windowFlags on each of them. If you cannot close one of them, please let me know.

chrherold, 2019-12-13:
3x no. Sorry :/

Running OSX 10.14.6 btw

commit 034ebb48 (2019-12-12): fix: disable close buttons of MDI subwindows on macOS (close #29)
commit 73b4d330 (2020-01-08): fix: subclass MDSubWindow and catch closeEvent (close #29)
closed 2020-01-08 by paulmueller