DC-analysis/DCscope #31: Analysis View/Plot - incorrect alignment and unwanted window rescaling

chrherold, 2019-12-11:
(Both Windows and Osx)

If the Plot section of the Analysis View is opened the elements are misaligned

[image]

Furthermore, when staying in this section and moving the window around or trying to rescale the window, it continuously expands in width for the time the window is clicked on.

Behavior in Event and Polygon Filter mode is correct. In these modes also the unwanted rescaling disappears but immediately reappears when changing back to Plot.

paulmueller, 2019-12-11:
I think I fixed it in e833ebc. Please reopen if the problem persists in 2.0.0b5.

chrherold, 2019-12-11:
The problem still persists in 2.0.0b5. Behavior is unchanged.

paulmueller, 2019-12-11:
I just updated my PyQt5 version and can reproduce it.

commit e833ebc3 (2019-12-11): fix: QuickView window resize incorrect (close #31)
commit cef917cf (2019-12-11): fix: statistics table widget kept growing in QuickView (close #31)
closed 2019-12-11 by paulmueller