DC-analysis/DCscope #33: Force subplots to have the same aspect ratio

paulmueller, 2020-01-08:
The aspect ratio is currently determined somehow by pyqtgraph taking the available space into account.

Pyqtgraph currently does not seem to have a feature that allows to make all plots the same size (such that if you overlay the plots, all axes and ticks would coincide).

chrherold, 2020-01-23:
one comment on that: subplot size seems to depend on the plot title

[image]

[image]

[image]

paulmueller, 2020-01-23:
yeah thanks, saw that too :roll_eyes:

paulmueller, 2021-02-23:
Currently we are using `pg.GraphicsLayoutWidget`. Maybe we should use [QGraphicsGridLayout](https://doc.qt.io/qt-5/qgraphicsgridlayout.html#details) - maybe with [setColumnFixedWidth](https://doc.qt.io/qt-5/qgraphicsgridlayout.html#setColumnFixedWidth).

paulmueller, 2021-02-24:
Workaround is to create an empty title label and anchor another label to it, see 616b71b.
See https://ci.appveyor.com/project/paulmueller/shapeout2/builds/37925366/artifacts for a preview.

closed 2021-02-24 by paulmueller