DC-analysis/DCscope #34: Cannot set filter ranges

felix-r, 2020-01-14:
I just updated to version b11 and now I cannot set the range of the box filters anymore. I can check the boxes but not edit the values.

[image]

paulmueller, 2020-01-14:
In the screenshot you are in the "box filter selection" mode. You have to hit "...Finish editing" before you can use the filters. Does the button work? If it's that, I guess it would make sense to highlight it in a different color or so.

felix-r, 2020-01-14:
Alright. That worked. Still have to get used to the new interface.

chrherold, 2020-01-14:
Paul was faster. :)

Yes either color, or we make a change to the optics on the filter selection mode and make numbers and bars invisible but give pictograms of the individual features instead. We pictograms in Shape-In. Although, inertia ratio will pose quite some challenge. 

[image]

paulmueller, 2020-01-16:
@chrherold Regarding pictograms: It might make sense, but as you said we don't have pictograms for all features. Please open a new issue if you think we should follow up on that.

commit 2a3a2b2c (2020-01-16): enh: be more verbose when user wants to add box filters (close #34)
closed 2020-01-16 by paulmueller