DC-analysis/DCscope #40: How to export the brightfield image from the dataset?

fyx18, 2020-05-18:
Hello, I think the software is very interesting, however, I didn't find the way to export the brightfield image from the dataset. Did the software haven't this function yet now?

paulmueller, 2020-05-18:
You should be able to export individual event images in Quick View by right-clicking on the event image. Is this enough for you or are you planning to export all event images?

fyx18, 2020-05-18:
Thank you for answering my question. I have exported the individual images successfully, actually I want to export all event images. Is there any way to do that?

paulmueller, 2020-05-18:
No, this is currently not possible with Shape-Out - I changed the label to enhancement.

May I ask why you need to export all event images to png files? Depending on your use case, generating thousands of png images only to import them in some other software might not be necessary.

fyx18, 2020-05-18:
I get it, thank you. 
Actually, I know this software from the article "Using real-time fluorescence and deformability cytometry and deep learning to transfer molecular specificity to label-free sorting". The article used the featureless AI-based classification, i.e., fed event images to the MLP, I think it's interesting and want to try the method in the article. However, I didn't have the image data, so I want to get the event images.

paulmueller, 2020-05-18:
The .rtdc files are just HDF5 files. The event images are located in the HDF5 Dataset located at "/events/image". You should be able to access them directly in the programming language of your choice (e.g. h5py for Python). If you are new to HDF5, I can recommend to you HDFViewer (to get an idea how .rtdc files are structured).

fyx18, 2020-05-18:
I'll have a try. Thank you for your help.

paulmueller, 2020-05-18:
At this point, I would also like to gently point out AIDeveloper (https://github.com/maikherbig/AIDeveloper) developed in our group and also used for RT-DC data. From what you said it might be exactly what you need.

closed 2020-05-18 by paulmueller