DC-analysis/DCscope #42: Image process

liang-minhui, 2020-06-08:
Hi all,
this is a beginner, I wonder if ShapeOut includes image processing? Or it's for processing the processed image data? and may I know which image processing code it uses,is it open source?

Appreciate any reply!
Liang

paulmueller, 2020-06-08:
I believe the documentation will answer your question: https://shapeout2.readthedocs.io/en/stable/

Shape-Out is a user interface for data analysis in real-time deformability cytometry (RT-DC). It is not a general-purpose image processing platform...

closed 2020-06-08 by paulmueller

liang-minhui, 2020-06-09:
May I ask some more questions?
If the image processing is done by Shape-In? Where can I download and install Shape-in?

Thank you!

chrherold, 2020-06-09:
The image processing and data acquisition tools you are asking about are part of the product sphere of Zellmechanik Dresden GmbH. 

https://www.zellmechanik.com/