DC-analysis/DCscope #46: Changing the orders of the plots

sdespina, 2020-07-28:
In Shape-out 1 the user has the option to change the order of the plots, this is not yet implemented in Shape out 2. I think it's important for having the correct layout/order in the exported svg file, especially when comparing 2 different groups with multiple repeats.

paulmueller, 2020-08-04:
What do you think - Should the order of the plots be governed by the order of the rows in the Block Matrix, or should each plot have its own ordering options (the latter is more difficult to implement)?

sdespina, 2020-08-04:
If one could move the order in the Block Matrix then I think this should suffice

felix-r, 2020-12-01:
After some time I wanted to create plots simply from ShapeOut2 and have to second Despina that it would be a really helpful feature. Especially if you plot with `marker hue = Dataset`, where you plot the data points from different datasets on top of each other, it is useful to set the order in which the datasets should be plotted.

paulmueller, 2020-12-01:
Thanks for your input. On second thought, I think I should give this issue higher priority.

commit d56e836a (2020-12-01): fix: allow to reorder the dataset list (close #46) and iconify buttons at the top of Analysis view
closed 2020-12-01 by paulmueller