DC-analysis/DCscope #48: Calculating YM from metadata

sdespina, 2020-08-06:
I have noticed that some discrepancies with the viscosity values when calculating the YM from metadata. I am a bit puzzled as I expect viscosity to decrease with increasing temperature.

I was using 0.6% Methyl cellulose. 
for example: 
1. 22.1°C 6.77mPA s
2. 22.7°C 9.88mPA s
3. 22.4°C 7.76mPA s
Please see the attach screen shots

[image]
[image]
[image]

paulmueller, 2020-08-06:
When I open a single measurement and edit the temperature, I see that the viscosity decreases with increasing temperature.
Maybe in these examples you have different channel sizes or flow rates? For CellCarrier B, the viscosity is computed according to equation 7 in https://arxiv.org/abs/1704.00572:
[image]

sdespina, 2020-08-07:
Ok thanks Paul! No this was the same size channel but you are right I have 3 different flow rates- I had a lot of conditions in this experiment and I thought it was the same flow rate but different concentration of cells (but in fact is the other way round!)
Sorry about the confusion!

closed 2020-08-16 by paulmueller