DC-analysis/DCscope #49: Not able to calculate Young's Modulus when Medium is set to "other"

chrherold, 2020-09-02:
When the medium property is "other" 
a) the temperature options cannot be selected
b) the viscosity cannot be entered (entering manually would be the only option for this property)

Accordingly, the Young's Modulus cannot be calculated.

[image]

paulmueller, 2020-09-02:
This is strange. The (a) behavior is intended. The (b) behavior works for me on Linux. I added a test function. Let's see what the outcome of the travis-CI macOS run is...
Could maybe send me the data file? Maybe this is some kind of special case.

paulmueller, 2020-09-02:
I believe this was caused by something like a race condition. There were two functions that update the accessibility of the temperature and viscosity spin controls depending on the input of the combo boxes. I combined these functions into one. So this should not happen again (tested on Windows). If 2.2.4 does not resolve this issue, please reopen.

closed 2020-09-02 by paulmueller