DC-analysis/DCscope #54: Event image background correction with "bg_image" feature.

paulmueller, 2020-10-23:
[M001_data.zip](https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/files/5430032/M001_data.zip)

maxschloegel, 2021-02-19:
I am on my way to implement the background correction and some things I am unsure about:

1. How exactly is the background correction supposed to be implemented? What is its exact purpose?

My first idea was to subtract the background from the cell-image and then take the absolute values. Problems:
- This leads inversion of the image (background is brighter→higher pixel values)
- Pixel inside the cell that are brighter than background are not inverted and therefore we lose information about pixel-variance in cell.

I could also invert this image to counteract the implicit inversion

If the background subtraction is for combating background drift, we could also compute the difference of "image_bg" to a reference "image_bg" (e.g. from the start of the data acquisition) and subtract this shift.

Which one is the right interpretation of "background correction"?

2. When implementing to hide the CheckBox when the dataset has not feature "image_bg", my current idea is to put the corresponding code into the setter-function of the "rtdc_ds"-property. I dont see any other similarly efficient location, but it feels a bit out of place in the setter function.
Maybe I need to understand the code better to find a more suitable location.
What is your opinion on that?

Thanks!

paulmueller, 2021-02-19:
1. Here, background correction is for visualization purposes only.
   I would not take the absolute value.
   - I did not anticipate image inversion. Maybe just subtract the background and add the mean of the background image to the entire array (Maybe this should only happen when "Auto contrast is set"?)?
   - Do you have example images? I imagine a background-corrected image to look like the original image without the channel visible (since it's only for visualization purposes).

   Here we are not dealing with background drift. This is could maybe become an ancillary feature at some point, but it is out of the scope of this issue.

2. Yes, I think the rtdc_ds.setter function would be the best candidate.

maxschloegel, 2021-02-19:
Re 1: 
Only subtracting the deviation from the mean of the background image solves the issue, good idea! It removes the channel very well. Here are some example images of the original and the corrected image with and without auto contrast:
No Background correction:
- No Auto Contrast
[image]
- With Auto Contrast
[image]

With Background Correction:
- No Auto Contrast
[image]
- With Auto Contrast
[image]

Is this what you had in mind?

I personally would keep the choices for "Auto Contrast" and "Subtract Background" independent, as their purpose is somewhat orthogonal, but I dont have a hard opinion on that and can add this condition.

Re 2: Ok, will do that!

paulmueller, 2021-02-19:
:+1: Sounds all good to me!

paulmueller, 2021-02-25:
fixed in https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/pull/78

closed 2021-02-25 by paulmueller