DC-analysis/DCscope #62: Not possible to make plots from .rtdc 2014 file from DCOR (no brightness issue?)

MartaUrb, 2020-11-27:
I tried to handle some very old RT-DC measurement from Guck Archive:
2014-11-26_1211_3ea1fd_v1.rtdc
https://dcor.mpl.mpg.de/dataset/2015-01-19_1241_44e50f

I can load it and do quick view, but I fail to generate any other plot with the error pasted below. Probably no brightness was recorded at that time and Shape-Out still tries to read it?

My System: macOS Catalina ver 10.15.7

Unhandled exception in Shape-Out version 2.4.0:
Traceback (most recent call last):
  File "shapeout2/gui/widgets/wait_cursor.py", line 31, in wrapper
  File "shapeout2/gui/main.py", line 269, in adopt_pipeline
  File "shapeout2/gui/analysis/ana_view.py", line 30, in adopt_pipeline
  File "shapeout2/gui/analysis/ana_view.py", line 41, in update_content
  File "shapeout2/gui/analysis/ana_plot.py", line 542, in update_content
  File "shapeout2/gui/analysis/ana_plot.py", line 178, in __setstate__
ValueError: 'bright_avg' is not in list

MartaUrb, 2020-11-27:
PS. It is possible to work with this file in ShapeOut 1.0.10

paulmueller, 2020-11-30:
Ok, this dataset also does not have any "image" and "contour" information from which "bright_avg" could be computed.

commit cf7681f3 (2020-11-30): fix: generating plots for datasets without bright_avg not possible (close #62)
closed 2020-11-30 by paulmueller