DC-analysis/DCscope #64: ShapeOut2 won't start on Windows

phidahl, 2020-12-03:
I tried installing the last 5 releases (Windows installer) of ShapeOut2

The same problem: Error Window showing: "Failed to execute Script ShapeOut2"

Installing with pip3 (Python38) didn't help either. By manually calling "python3.exe Scripts/shapout2.exe" I get the following error:

```
C:\Users\ZMD\AppData\Local\Programs\Python\Python38>python.exe Scripts\shapeout2.exe
 ** On entry to DGEBAL parameter number  3 had an illegal value
 ** On entry to DGEHRD  parameter number  2 had an illegal value
 ** On entry to DORGHR DORGQR parameter number  2 had an illegal value
 ** On entry to DHSEQR parameter number  4 had an illegal value
Traceback (most recent call last):
  File "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "Scripts\shapeout2.exe\__main__.py", line 7, in <module>
  File "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\lib\site-packages\shapeout2\__main__.py", line 9, in main
    import pyqtgraph  # noqa: F401
  File "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\lib\site-packages\pyqtgraph\__init__.py", line 19, in <module>
    import numpy  ## pyqtgraph requires numpy
  File "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\lib\site-packages\numpy\__init__.py", line 305, in <module>
    _win_os_check()
  File "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\lib\site-packages\numpy\__init__.py", line 302, in _win_os_check
    raise RuntimeError(msg.format(__file__)) from None
RuntimeError: The current Numpy installation ('C:\\Users\\ZMD\\AppData\\Local\\Programs\\Python\\Python38\\lib\\site-packages\\numpy\\__init__.py') fails to pass a sanity check due to a bug in the windows runtime. See this issue for more information: https://tinyurl.com/y3dm3h86
```

There seems to be a problem with numpy and windows runtime in version 2004. Is there a workaround? Or should we maybe just wait a bit? - I already installed an update for windows to version 20H2 which didn't help.

P.S.

I installed an older version of numpy: `pip3 install numpy==1.19.3` and with that it works. But the bug is still there, I guess. The question is, if ShapeOut2 is affected by it or not.

P.P.S 

The pip version of ShapeOut2 only works if I start it with
```
C:\Users\ZMD>cd C:\Users\ZMD\AppData\Local\Programs\Python\Python38

C:\Users\ZMD\AppData\Local\Programs\Python\Python38>python2 Scripts\shapeout2.exe
```

simple `shapeout2` results in:

```
 ---------------------------
Fatal Error in Launcher
---------------------------
Unable to create process using '"c:\users\zmd\appdata\local\programs\python\python38\pythonw3.exe"  "C:\Users\ZMD\AppData\Local\Programs\Python\Python38\Scripts\shapeout2.exe" ': Das System kann die angegebene Datei nicht finden.

---------------------------
OK   
---------------------------
```

paulmueller, 2020-12-03:
Thanks for reporting!

I don't really understand why you seem to be the only one with this issue. Maybe it is because you have Python installed alongside the Shape-Out 2 installation? That would then be a PyInstaller bug. What's your Windows version? It should work with 10.0.18363 and 10.0.14393...

Regarding the subsequent errors:

1. https://stackoverflow.com/questions/64654805/how-do-you-fix-runtimeerror-package-fails-to-pass-a-sanity-check-for-numpy-an
indeed suggests that this is related to numpy 1.19.4.

2. How can this work: `python2 Scripts\shapeout2.exe`? (Shape-Out 2 is Python3-only)

3. And I don't know why `shapeout2` should not work. Do the files `c:\users\zmd\appdata\local\programs\python\python38\pythonw3.exe` and `C:\Users\ZMD\AppData\Local\Programs\Python\Python38\Scripts\shapeout2.exe` not exist? I currently cannot debug this, because I am still waiting for my new laptop where I can run Windows in a virtual machine.

paulmueller, 2020-12-03:
I pinned the numpy version in this build: https://ci.appveyor.com/project/paulmueller/shapeout2/builds/36639495/artifacts
Could you please check whether it works now?

paulmueller, 2020-12-03:
Alternatively, I bumped PyInstaller to 4.1 in this build: https://ci.appveyor.com/project/paulmueller/shapeout2/builds/36639586/artifacts
If that works, than I would prefer to keep it like that.

paulmueller, 2020-12-03:
According to https://developercommunity.visualstudio.com/content/problem/1207405/fmod-after-an-update-to-windows-2004-is-causing-a.html this is caused by a problem with the Windows DLL ucrtbase.dll.

phidahl, 2020-12-03:
>  How can this work: python2 Scripts\shapeout2.exe? (Shape-Out 2 is Python3-only)
It was of course python3

As mentioned the windows versions were 2004 and 20H2

> And I don't know why shapeout2 should not work. Do the files c:\users\zmd\appdata\local\programs\python\python38\pythonw3.exe and [...]

The files do both exist. That is strange.. I'm going to try the builds and report back.

About the bad windows DLL: Let's hope that we don't use the broken functions.

phidahl, 2020-12-03:
> Alternatively, I bumped PyInstaller to 4.1 in this build: https://ci.appveyor.com/project/paulmueller/shapeout2/builds/36639586/artifacts
If that works as well, then I would prefer to keep it like that.

This build worked. Thanks!

commit 02465704 (2020-12-03): update changelog (close #64)
closed 2020-12-03 by paulmueller