DC-analysis/DCscope #67: Adding new filter automatically selects Box filters from previously edited filter

B-Hartmann, 2020-12-07:
When I add a new filter, the selected options ("Box filters") from the filter that I edited before are already selected.
For example:
My first filter is for fluorescence and I select FL-1 width and FL-1 maximum. When I add another filter for porosity, FL-1 width and FL-1 maximum are already pre-selected. I then add another filter for brightness and porosity is preselected. Furthermore, I noticed that when I now go back to edit the first filter for fluorescence and change for example the lower limit of FL-1 maximum, then press "Apply", the brightness is selected as "Box filters" there as well.
Is this behavior intended?

paulmueller, 2020-12-07:
No, this sounds like a bug to me.

paulmueller, 2021-01-04:
fixed in 914586263728aaca7b373729477c41e15b843e1b

closed 2021-01-04 by paulmueller