DC-analysis/DCscope #70: macOS builds won't run on Catalin 10.15.7

paulmueller, 2021-01-11:
@MartaUrb reported that the older release (https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/releases/tag/2.4.3) won't run anymore on macOS 10.15.7 (no visual feedback at all).

@phidahl Can you reproduce this? If yes, could you please check if there is any output if you start it from the command line?

I am referring to the 2.4.3 build, because the new 2.4.14 build might not work because it was built on an entirely new build system.

phidahl, 2021-01-12:
Hi guys,

I’m using not Catalina but BigSur.

I just tried it. On starting I get this error:

"You are using an unexpected version of pyqtgraph. Plot export may not work correctly!“

The program seems to work anyway.

When I click on an event (Quick View) in a dataset from dcor I get this:
"Unhandled exception in Shape-Out version 2.4.14:
Traceback (most recent call last):
  File "/Applications/Shape-Out.app/Contents/MacOS/shapeout2/gui/quick_view/qv_scatter.py", line 252, in set_point
    self.sigClicked.emit(self, self.ptClicked)
TypeError: ScatterPlotItem.sigClicked[object, object, object] signal has 3 argument(s) but 2 provided“

Hope this helps!

Cheers
Philipp

MartaUrb, 2021-01-12:
Hi there,

I guess this was a false alarm after all!

As I mentioned already yesterday to Paul I can run the 2.4.3 version but it takes roughly 20 minutes before the Shape-Out loads. After that it works as it used to.

I updated now to 2.4.14 and it works as well (also with a very long time to open the software). Similar to Philip I am getting this error when opening the software:
"You are using an unexpected version of pyqtgraph. Plot export may not work correctly!“
show details says:
"If you have installed Shape-Out using pip, please install our pyqtgraph fork using: 
pip install git+git://github.com/ZELLMECHANIK-DRESDEN/pyqtgraph.git"
- I haven't installed using pip

I could load and plot data with the new version regardless. I could also do quickview without any error prompt.

I made a screen shot of my activity monitor with top-scorers of CPU usage during ShapeOut opening. The "kav" process drops to below 1% afterwards. Kav stands for Kaspersky AntiVirus - our new anti-virus software.... I am not sure why it should act up like this?

Cheers,

Marta

[image]

paulmueller, 2021-01-13:
It looks like Kaspersky is checking the executable before it can start. A quick search through the PyInstaller issue pages did not reveal anything.

@MartaUrb One option might be to get into contact with MPL IT. Maybe they can add Shape-Out to a "white-list", so Kaspersky does not check it. If MPL has a contract with Kaspersky, then maybe this can be resolved via customer support...

MartaUrb, 2021-01-14:
ok, I have sent a trouble ticket to the IT at MPL (and cc-ed you Paul :) )

paulmueller, 2021-02-23:
Looks like this has been fixed by someone else.

closed 2021-02-23 by paulmueller