DC-analysis/DCscope #71: Add support for new emodulus LUTs

paulmueller, 2021-01-18:
- [ ] get rid of the deprecation warnings from dclab 0.32.0
- [ ] this affects saved sessions
- [ ] allow to set LUT identifier for emodulus computation
- [ ] allow to set LUT identifier for isoelastics (QuickView)
- [ ] allow to set LUT identifier for isoelastics (Plots)

RaghavaAlajangi, 2024-09-17:
Hi @paulmueller,
I think I understand the first point differently. I thought I just needed to write a test function to make sure old session files load with LUT specified when we open with shapeout2 but there is already a relevant test function for that.

did you mean to implement a way that old session files get the LUT in their metadata when we open them with shapeout2 and then the LUT gets saved automatically in that session file when we close them?

paulmueller, 2024-09-17:
:thinking: It could be that I already implemented that test. Which test function are you referring to?
No, I would not modify any existing session files.

RaghavaAlajangi, 2024-09-18:
```
def test_2_5_1_replace_emodulus_model():
    """In Shape-Out 2.5.1 we replace "emodulus model" with "emodulus lut"."""
    spath = pathlib.Path(__file__).parent / "data" / "version_2_1_0_basic.so2"
    pl = session.open_session(spath)
    sc = pl.slots[0].config
    assert "emodulus" in sc
    assert "emodulus model" not in sc["emodulus"]
    assert "emodulus lut" in sc["emodulus"]
    assert sc["emodulus"]["emodulus lut"] == "LE-2D-FEM-19"
```
This test function is not intended exclusively for the above use case. But, it is being tested!

paulmueller, 2024-09-18:
Thanks, yes that is already the test I was looking for.

paulmueller, 2024-11-12:
Implemented in 2.19.0 as per the PRs listed above.

closed 2024-11-12 by paulmueller