DC-analysis/DCscope #82: Plot name has problems with less-than sign <

B-Hartmann, 2021-03-19:
When using the less-than sign "<" in the plot name, the string is not shown correctly in the plot window, it gets clipped directly before the "<". The plot window shows the complete name, though, it's just the string within the window.
No problems with the greater-than sign ">".

ShapeOut 2.6.0
[image]

paulmueller, 2021-03-19:
Interesting. This looks like a pyqtgraph issue and should be fixed upstream. Will create an issue there when I have time. In the mean time, you could use "≤".

paulmueller, 2021-04-29:
Upstream issue: https://github.com/pyqtgraph/pyqtgraph/issues/1750

paulmueller, 2021-04-30:
Turns out this is not a bug, but a feature. Pyqtgraph interprets those strings as html. I will disable this in the next release.

commit 508bc8ab (2021-04-30): fix: HTML-escape plot titles and axis labels (close #82)
closed 2021-04-30 by paulmueller