DC-analysis/DCscope #83: Background correction leads to white line in center of channel

marketakub, 2021-04-07:
Measurements of blood (sample fluid prepared as usual 50 ul blood + 950 ul 0.6% MC) at the clinics are now done with continuous background storage, with the image_bg feature. Without background correction the images look OK, but when BGR correction is applied there is a white line in the center.

[image]
[image]
[image]
[image]

marketakub, 2021-04-07:
Small sample dataset
[sample.zip](https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/files/6272528/sample.zip)

marketakub, 2021-04-07:
Small sample dataset
[sample.zip](https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/files/6272553/sample.zip)

paulmueller, 2021-04-07:
To me it looks like the images from which the background image `image_bg` is computed contain cells. I.e. Shape-In does not detect cells in some images, assumes that they don't contain cells and uses those images for background computation. The background images vary a lot in-between events.

[image]