DC-analysis/DCscope #85: Support dclab plugin features in Shape-Out2

PinkShnack, 2021-05-18:
Plugin features are a way for users to define their own features within dclab. The original dclab issue is [here](https://github.com/ZELLMECHANIK-DRESDEN/dclab/issues/105) and the soon to be merged PR is [here](https://github.com/ZELLMECHANIK-DRESDEN/dclab/pull/113).

These should be available in Shape-Out2.

maikherbig, 2021-07-28:
I would be highly interested in that feature!!!

paulmueller, 2021-07-28:
Thanks! We are currently very busy with other projects, so don't expect this to be implemented within the next month.

paulmueller, 2021-10-15:
Before this can be implemented, we need https://github.com/ZELLMECHANIK-DRESDEN/dclab/issues/144 and https://github.com/ZELLMECHANIK-DRESDEN/dclab/issues/143.

paulmueller, 2021-11-24:
TODO:

- [ ] tests
- [ ] docs

paulmueller, 2021-11-26:
See https://shapeout2.readthedocs.io/en/stable/sec_qg_extensions.html

closed 2021-11-26 by paulmueller

PinkShnack, 2021-11-26:
Fantastic! Thanks for doing this Paul!