DC-analysis/DCscope #88: Open session files by double click

phidahl, 2021-06-08:
Many programs are linked to certain file types in the os, so that double clicking a document files for example opens libreoffice with the clicked document. This intuitively integrates a program into a user's workflow.

Would that be possible for ShapeOut, too? - I think the implementation might differ on the platforms, but Windows should be most common and have highest priority.

Alternatively or additionally drag & drop of session files could open the session. Dropping data files is working on all platforms which is already very handy.

paulmueller, 2021-06-08:
Yes, that would have to be done in the InnoSetup configuration file.

paulmueller, 2021-06-27:
Fixed by the commits above.

closed 2021-06-27 by paulmueller