DC-analysis/DCscope #89: Statistical significance calculation fails

Ingogan, 2021-06-22:
When trying to perform statistical significance calculation fails, the following error message appears:

Unhandled exception in Shape-Out version 2.6.4:
Traceback (most recent call last):
  File "shapeout2\gui\compute\comp_lme4.py", line 89, in on_lme4
    rlme4 = lme4.Rlme4(model=self.model, feature=self.feature)
  File "dclab\lme4\wrapr.py", line 52, in __init__
  File "dclab\lme4\rsetup.py", line 160, in install_lme4
  File "rpy2\robjects\functions.py", line 198, in __call__
  File "rpy2\robjects\functions.py", line 125, in __call__
  File "rpy2\rinterface_lib\conversion.py", line 45, in _
  File "rpy2\rinterface.py", line 677, in __call__
rpy2.rinterface_lib.embedded.RRuntimeError: Error in (function (pkgs, lib, repos = getOption("repos"), contriburl = contrib.url(repos,  : 
  unable to install packages

paulmueller, 2021-06-22:
Thanks for reporting this issue. Could you please answer the following questions for me:

How did you install Shape-Out 2?

How did you install R?

Which operating system are you using (Windows?)?

Ingogan, 2021-06-23:
Dear Paul,

I am using Windows 10 Professional 21H1 (Build 19043.1055).
I downloaded the installation exe files and followed the protocol.

Best regards,

Ingo

paulmueller, 2021-06-23:
To me it looks like you don't have the permission to install the "lme4" and the "statmod" packages. Could you please try uninstalling R and reinstalling it? You could also try to install these packages manually from within R (There are a multitude of help pages that show you how it is done, e.g. [this one](https://www.datacamp.com/community/tutorials/r-packages-guide)).

paulmueller, 2021-06-25:
I just released version 2.6.5 which is also built on top of a newer version of dclab. There was a fix in dclab that also involved the installation of R packages. Could you please check if it already works for you now?

Ingogan, 2021-06-26:
Dear Paul,

I have reinstalled R 4.1 and Shape-Out 2.6.5. Now I have a new message when opening the software:
"You are using an unexpected version pyqtgraph. Plotting may not work correctly!
If you have installed Shape-Out using pip, please install our pyqtgraph fork via: 
pip install pyqtgraph==0.12.0"

Statistical analysis yields:
"Unhandled exception in Shape-Out version 2.6.5:
Traceback (most recent call last):
  File "shapeout2\gui\compute\comp_lme4.py", line 89, in on_lme4
    rlme4 = lme4.Rlme4(model=self.model, feature=self.feature)
  File "dclab\lme4\wrapr.py", line 52, in __init__
  File "dclab\lme4\rsetup.py", line 163, in install_lme4
  File "rpy2\robjects\functions.py", line 198, in __call__
  File "rpy2\robjects\functions.py", line 125, in __call__
  File "rpy2\rinterface_lib\conversion.py", line 45, in _
  File "rpy2\rinterface.py", line 677, in __call__
rpy2.rinterface_lib.embedded.RRuntimeError: Error in (function (pkgs, lib, repos = getOption("repos"), contriburl = contrib.url(repos,  : 
  unable to install packages"

Should I try the package installation as you previously suggested or something else?

Best regards

paulmueller, 2021-06-26:
Sorry about the first message. That is a false alarm.

I see you are using R-4.1.

Could you please retry with R-3.6.3?

Ingogan, 2021-06-27:
Using R-3.6.3 it now downloads something doing the calculations and gives the following error message:

Unhandled exception in Shape-Out version 2.6.5:
Traceback (most recent call last):
  File "shapeout2\gui\compute\comp_lme4.py", line 91, in on_lme4
    wds.add_to_rlme4(self.pipeline, rlme4)
  File "shapeout2\gui\compute\comp_lme4_dataset.py", line 50, in add_to_rlme4
    rlme4.add_dataset(ds=ds, group=group, repetition=repetition)
  File "dclab\lme4\wrapr.py", line 84, in add_dataset
ValueError: A dataset with group 'control', repetition '1', and 'channel' region has already been added (index 0)!

paulmueller, 2021-06-27:
OK, thanks for testing. This means that the original issue can be resolved by downgrading R to 3.6.3.

Regarding the new error message, it seems like you gave two datasets the same repetition in the control group. Could you please check whether that is the case? Here is a link to the relevant section in the docs: https://shapeout2.readthedocs.io/en/stable/sec_qg_lme4.html

paulmueller, 2021-07-07:
I assume that this was a permissions problem. If anyone encounters this in the future, one of these options may also help:
- Reinstall R for the current user only (not system-wide)
- Run RStudio as administrator and install the packages "lme4" and "statmod"
- https://stackoverflow.com/questions/42807247/installing-package-cannot-open-file-permission-denied/50787084#50787084

closed 2021-07-07 by paulmueller

Ingogan, 2021-07-10:
Dear Paul,

thank you so much for solving the problem!