DC-analysis/DCscope #95: Feature "index" is missing after exporting

maikherbig, 2021-08-06:
ShapeOut2 version: 2.6.8
Procdure to reproduce:
- Load RT-FDC dataset
- Export - Data
- Radiobutton: RT-DC (*.rtdc)
- select all features
- OK

When checking the resulting file (e.g.) using HDFview, the feaure "index" is suddenly missing. 
However, I need that property as I do filtering outside of ShapeOut and I would like to keep track which cells from the original file are kept

paulmueller, 2021-08-06:
I believe what you need is `index_online`. `index` always enumerates the dataset and must not have gaps in the enumeration, because it is used by hierarchy children. If it would be exported, it would just be `np.arange(len(dataset))`. See also https://github.com/ZellMechanik-Dresden/dclab/issues/71.

maikherbig, 2021-08-09:
Thanks, that does exactly what I want :)

closed 2021-08-09 by maikherbig