DC-analysis/DCscope #97: Enhancement idea: action to remove all measurements in a session/save session templates

felix-r, 2021-08-20:
I currently had a use case where I would use the very same filters and plots for different sets of datasets in different ShapeOut sessions. With the "clear session" button, all filters and plots are also removed but for me, I only wanted to remove the datasets and load new datasets to use the already defined filters and plots. For this, a button to remove all datasets would be helpful or even handier: if you could save the session in a template format without the datasets.

Just a suggestion, nothing urgent. Cheers!

paulmueller, 2021-08-20:
You can already save your filters via the "Export" menu. Saving plots (creating the templates you described) would be a major effort. You have to map datasets to plots which is not trivial, so you need a new interface that asks the user what should go where.

felix-r, 2021-08-20:
But I have the feeling what I like to have works almost out of the box already: I work on a session creating plots and filters for a set of datasets. When I'm done, I remove all datasets, load the new datasets I want and reactivate the filters and plots which are already  defined in the session.

If saving a template is too complex, no problem! What a about a button to remove all datasets from the session?

DerDeef, 2021-12-02:
Hi Paul,
I have the same suggestion/wish: instead of removing every single dataset, it would be great to be able to remove all at once. The filter and plotsettings remain and it is possible to import another batch of datasets. To easily save those settings it looks like you can safe a session were you removed all datasets. Worked for me, in a quick test.

commit 22d4a75f (2021-12-08): enh: add action to remove all datasets from a session (close #97)
closed 2021-12-08 by paulmueller