DC-analysis/DCscope #98: Shapeout_2.7.0 Failed to execute Script

sdespina, 2021-08-24:
I installed the latest version but I can not run it on my computer. 
[image]

paulmueller, 2021-08-24:
Can anyone else reproduce this?

paulmueller, 2021-08-24:
@sdespina But the previous versions of Shape-Out worked, right?

sdespina, 2021-08-24:
> @sdespina But the previous versions of Shape-Out (e.g. 2.6.9) worked on that computer, right?

Yes, i already reinstalled version 2.6.6. that I had before and works fine. I didn't try the other ones, cos I wanted to continue with my analysis. I can give it a try tomorrow

paulmueller, 2021-09-13:
This issue is also related to #89. In the command line, you might see errors like this one:
[image]

This is also an issue with Windows support by rpy2: https://github.com/rpy2/rpy2/issues/721

A workaround for Windows users is to install R-3.6.3: https://cran.r-project.org/bin/windows/base/old/3.6.3/

I am closing this issue here (because it is not a Shape-Out bug), but I am pinning it until the upstream issue is eventually resolved.

closed 2021-09-13 by paulmueller