DC-analysis/DCscope #100: Missing option to delete polygon filter

B-Hartmann, 2021-08-30:
The option to delete a polygon filter is missing, which is unfortunate if you miss-clicked and accidentally created a new one. Would be nice to have a third button in the polygon filter menu in the quick view window to delete it..

paulmueller, 2021-08-31:
It should only be possible to delete a polygon filter if it is not part of a filter ray.

B-Hartmann, 2021-09-01:
I just realized by trying out that unused polygon filters are not saved when you use the option "Save Session". I created a new polygon filter, selected it in a filter, clicked apply, then deselected the polygon filter, saved the session, closed Shape-Out2, started it again, loaded the session, and it was not there.

paulmueller, 2021-09-01:
OK, that makes sense to me. Does it make sense to you? (At least it's a workaround for now) I think an option to delete a polygon filter might still improve the user experience.

B-Hartmann, 2021-09-02:
It is a workaround for now, yes.
But I sometimes activate or deactivate filters when analyzing the data, and when I save a session, I expect that all the filters and plots that I created are saved, even the unused ones.

paulmueller, 2021-09-02:
I created a new issue for the session-saving: https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/issues/101