DC-analysis/DCscope #110: Visualization bugs in Plot: Contour position dependent on spacing resolution

chrherold, 2021-11-30:
found the following unexpected behavior in Shape-Out 2.8.1:

The position of the contour lines seem not to match the density plot at all at some settings for spacing

[image]

while they match the density plot for other spacing settings

[image]

I can imagine that the spacing resolution influences the contour accuracy but I believe it should not completely mismatch
the contour position if the underlying density map has its maximum clearly elsewhere.

paulmueller, 2021-11-30:
It certainly looks strange. Can I access that dataset somewhere? I could not reproduce it on my machine.

paulmueller, 2021-11-30:
The thing is that the density map you see in the scatter plot is not identical to the density map used for contour computation.
Also note that Shape-Out computes the contour lines based on percentiles (events within a contour line). The artifacts that you are seeing are due to extremely large bins in the histogram binning method. Here are a few examples that illustrate this:

[image]

[image]

[image]

Basically, whenever you have "kinks" in your contour line, your spacing is too large.

I have the feeling that it might be reasonable to automatically refine the spacing until there are no more kinks at the expense of the user having to wait longer for their plots.

@chrherold  What do you think?

paulmueller, 2021-12-01:
As discussed,

- a temporary solution to this issue is a button that can automatically refine the contour spacing based on kinks and a warning in the plot that states that the contour spacing is too large
- an ultimate solution based on a new implementation in dclab: https://github.com/ZELLMECHANIK-DRESDEN/dclab/issues/153

paulmueller, 2021-12-05:
Bullet point 1 in my previous post is resolved with ebe507a and 88ab772.

Here is an example (note the red text):

[image]

And there is a new button in the plot tab for setting "sane" values to the contour spacing:

[image]

@chrherold Please have a look at the development release here and let me know whether we can incorporate this temporary solution: https://ci.appveyor.com/project/paulmueller/shapeout2/builds/41778981/artifacts

If this works well enough, I would close this issue here. Regarding the second bullet point, see issue #111.

chrherold, 2021-12-08:
I tested the temporary solution and it works very well. Especially the find best spacing button does a very good job.

The indicator in the plot seems to move around a bit weirdly, if there is more than one row

[image]

And it shows multiple times if "One plot per dataset" is selected. Maybe it could be placed inside the plotting panels that contain a contour drawing. This way there would also not be some cutting out a part of the figure around it.

paulmueller, 2021-12-08:
Thanks for testing.
Please try the latest dev version: https://ci.appveyor.com/project/paulmueller/shapeout2/builds/41818695/artifacts
[image]

chrherold, 2021-12-10:
Works very well! Thanks!

closed 2021-12-10 by paulmueller