DC-analysis/DCscope #114: Edit->Preferences causes error

maikherbig, 2022-01-03:
ShapeOut2 2.9.2
How to produce the error: click Edit, click Preferences.
(maybe related: I earlier switched on Developer mode in the preferences.)
Error message:
Unhandled exception in Shape-Out version 2.9.2:
Unhandled exception in Shape-Out version 2.9.2:
Traceback (most recent call last):
  File "shapeout2\gui\main.py", line 714, in on_action_preferences
    dlg = preferences.Preferences(self)
  File "shapeout2\gui\preferences.py", line 76, in __init__
    self.reload()
  File "shapeout2\gui\preferences.py", line 117, in reload
    self.reload_lme4()
  File "shapeout2\gui\widgets\wait_cursor.py", line 31, in wrapper
    ret = func(*args, **kwargs)
  File "shapeout2\gui\preferences.py", line 156, in reload_lme4
    rsetup.set_r_path(binary)
  File "dclab\lme4\rsetup.py", line 177, in set_r_path
  File "dclab\lme4\rlibs.py", line 42, in import_r_submodules
  File "importlib\__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 476, in exec_module
  File "rpy2\robjects\__init__.py", line 20, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 476, in exec_module
  File "rpy2\robjects\functions.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 476, in exec_module
  File "rpy2\robjects\help.py", line 46, in <module>
  File "rpy2\robjects\help.py", line 41, in quiet_require
  File "rpy2\rinterface_lib\conversion.py", line 45, in _
  File "rpy2\rinterface.py", line 103, in parse
  File "rpy2\rinterface_lib\_rinterface_capi.py", line 640, in _parse
ffi.error: symbol 'R_tryCatchError' not found in library 'C:\Users\MaikH\AppData\Local\Shape-Out\R\R-3.2.5\bin\x64\R.dll': error 0x7f

paulmueller, 2022-01-04:
Thanks for reporting. Could you please check whether installing R 3.6.3 resolves this issue?

https://cran.r-project.org/bin/windows/base/old/3.6.3/

paulmueller, 2022-01-12:
I just noticed that you are using the R binary from Shape-Out 1. A quick solution would be to uninstall Shape-Out 1 (or rename the path ``C:\Users\MaikH\AppData\Local\Shape-Out\R\R-3.2.5``.

commit 5c44d8c2 (2022-01-12): fix: handle unsupported R versions properly (close #114)
closed 2022-01-12 by paulmueller