DC-analysis/DCscope #117: Plugin features/extensions do not show up initially in list of features when exporting statistics

B-Hartmann, 2022-02-16:
Shape-Out 2.9.4
Windows 10

When I have a dataset and some filters and want to compute statistics, the list of features doesn't show my extensions/plugin features. I have to open the preferences once (not even opening the tab "Extensions") and then go to "export statistics" again to see all features and be able to select them.
It's like Shape-Out forgot that I added the extensions and needs a reminder first...

paulmueller, 2022-02-16:
I cannot reproduce it with 2.9.5 on Ubuntu using the fluorescence density example plugin feature from https://shapeout2.readthedocs.io/en/stable/sec_qg_extensions.html#example-fluorescence-volume. Could you please try again?

B-Hartmann, 2022-02-17:
Shape-Out 2.9.5 doesn't fix the issue.
I cannot see the added features anywhere, not in the drop-down menus in the quickview, not in the drop-down menus in plot settings, not in the list for calculating statistics.
I need to open the preferences first and close it agian, then select a dataset again in the quickview and closing and re-opening the analysis view to update all the drop-down menus.

B-Hartmann, 2022-02-17:
Btw, in the first blue info box on the documentation website you linked above, it says _If you installed Shape-In via installer (not via pip), then many extensions might not work ...._  
Is it correct that it says Shape-In here? Should it not be Shape-Out?

commit 5d918bff (2022-02-17): fix: extensions not loaded during startup on Windows (close #117)
closed 2022-02-17 by paulmueller