DC-analysis/DCscope #120: Docs don't explain fluorescence crosstalk compensation

B-Hartmann, 2022-02-24:
In the docs, there is just half a sentence under "User interface - Analysis view" where the compensation matrix for crosstalk is mentioned, yet there is no explanation on how to use it or how it works. An example would be nice. Maybe also an option to use offsets per FL channel.

B-Hartmann, 2022-02-24:
If I compensate for crosstalk, some events will get a negative FL-X value. Since fluorescence signals are usually looked at with logarithmic axis scales, these events disappear, and I see less events than I really measured. This would be something I would like to have in the docs, maybe with a guide how to handle the data correctly in that case.

paulmueller, 2022-02-28:
@phidahl What is your opinion? How do you normally deal with **offsets** in the fluorescence signal?

@B-Hartmann  Updating the docs should not be a problem. It would probably make sense to add a quick guide for that. Do you have nice, published datasets that I could use?

@B-Hartmann The negative values for fluorescence data is a known issue (see https://github.com/ZELLMECHANIK-DRESDEN/dclab/issues/56). This is not going to be implemented soon.

phidahl, 2022-02-28:
As far as I know in other cytometry software it is common to compensate with the aim that a negative population has a mean value of 0. Then you have to live with negative values.
It would be best, if this could be plotted with an appropriate scale. If this is not possible the matrix could be modified to add an offset large enough to have all events in the plottable range.

Device-wise there should not be any negative Fl-max values if the setup ist correctly adjusted and in order.