DC-analysis/DCscope #121: Cannot select crosstalk-corrected FL signal for plotting

B-Hartmann, 2022-02-24:
Shape-Out 2.9.6
Windows 10

I use the fluorescence crosstalk compensation matrix and want to plot the fluorescence signals in a plot. In quickview, I can choose "FL-1 maximum, crosstalk-corrected [a.u.]", but in the plot, this feature does not show up.  
It seems that I can only select it in the plot's options, if the crosstalk matrix was changed for every single dataset that I want to have in that plot. So if I only need compensation in one of the datasets, I would need to also change the matrices of all the other datasets, for example, by just by the minimum possible value, because only then this feature exists for these datasets and it can be choosen as an axis for the plot.

B-Hartmann, 2022-02-24:
It would be nice if I could choose the feature once I compensated in at least one dataset regardless of any changes in the crosstalk compensation matrix in other datasets. If all entries are zero in the matrix, it could just plot the normal FL signal.

paulmueller, 2022-02-24:
I strongly disagree with this feature request, because it potentially produces plots that are misinterpreted. By design, the axes are shared between all plots. I would propose an alternative, where the user has a button "no crosstalk" for each dataset, where he can set the matrix to unity. This would at least reduce the number of clicks. Another option for you would be to creat two separate plots?

B-Hartmann, 2022-02-24:
I think there is no need for a button "no crosstalk", since I can just select the other "FL-X maximum" feature in the plot settings. I assume that this button would reset all entries in the matrix, which I would need to type in again if needed. 
Creating several plots and manually set the same axis limits for comparison etc. would work and seems the best solution here.

paulmueller, 2022-02-24:
OK great!

closed 2022-02-24 by paulmueller