DC-analysis/DCscope #123: Filtering for one "Most probable ML class" not possible with one box filter

B-Hartmann, 2022-05-17:
ShapeOut 2.11.0

When using a box filter to only get events with the same "most probable ml class", let's say, ml_class = 4, I run into the following problem:
The box filter for that feature only allows integer values, and when I filter with min=4 and max=4, the filter does not work at all, but if I filter with min=3 and max=4, both classes, 3 and 4, are selected, although I only want ml_class = 4. 

The current workaround is to use two filters, one with min=3 and max=4 and the other one with min=4 and max=5.
I think it would be nice if the box filter would work by setting min and max value of the filter to the same value (4 in my example).

B-Hartmann, 2022-05-18:
I am not sure I can follow your thoughts. If the boundaries were included, that would be exactly what I wanted. Why would the behavior change if the bug was fixed?

paulmueller, 2022-05-18:
If you set min=3 and max=4 and you get both classes, then the boundaries (min and max) are included. My initial thought was that one of the boundaries should not be included (so you only get one class), but that might break other pipelines.

commit 6cc715bd (2022-05-18): fix: allow to specify 0.5 increments when filtering integer features (close #123)
closed 2022-05-18 by paulmueller