DC-analysis/DCscope #124: Young's modulus calculation for different buffer

B-Hartmann, 2022-06-09:
Shape-Out 2.11.2
Windows 10

I have measurements where the temperature was not recorded or falsely measured to be 0.0°C and the buffer was selected incorrectly in Shape-In, so I cannot calculate the Young's modulus right now. The buffer is a custom one with 0.84 % methylcellulose, so I would need to change the viscosity manually.
When doing the measurement, "CellCarrier B" was selected in Shape-In, although another buffer was used.

In the "Dataset" tab in the analysis view, I cannot change the medium and I cannot change the viscosity of the buffer. This can be helpful to avoid errors when all the necessary information needed to calculate Young's modulus are there, but prevents me right now from fixing the problem.

Using the "Bulk action" --> compute Young's modulus method does not seem to work, because after entering all the information (medium = other, viscosity = 6.73887 mPa *s (value for my buffer as measured with a rheometer) and clicking ok, nothing happens. If I open the menu to set the parameters for the Young's modulus' calculation again, it looks like I didn't change anything, so the old (initial) values are shown there. And when exporting the data, always the same values for the Young's modulus are exported, no matter what I enter in the "Bulk action" menu.

Can it be that Shape-Out does not allow to change these settings when a valid medium (CellCarrier or CellCarrier B) was selected in Shape-In?

B-Hartmann, 2022-06-09:
I would have expected that when I use the "Bulk action" method, the buffer, temperature and viscosity for all datasets opened in my session will temporarily be overridden in the session and the Young's modulus will be computed based on the values I entered. And these values for Young's modulus would then be exported when using the "Compute"-->"Statistical data option.

paulmueller, 2022-06-10:
Shape-Out uses dclab under the hood and dclab fully trusts the raw measurement (meta-)data. The reason for not being able to change buffers or other meta data is to prevent users from doing a bad/wrong analysis. The only option you have is to remove the temperature feature from the dataset by manually editing the files using e.g. HDFView or Python+h5py.

paulmueller, 2022-06-10:
However, what Shape-Out could at least do is to tell the user that it did not analyze according to the user meta data, but according to the metadata in the .rtdc files.

B-Hartmann, 2022-06-10:
> The reason for not being able to change buffers or other meta data is to prevent users from doing a bad/wrong analysis.

That's what I thought. I would say the [docs](https://shapeout2.readthedocs.io/en/stable/sec_qg_youngs_modulus.html) need a little update then, because to me it looks (based on the text and especially the images) like one should be able to set the buffer and temperature freely in the "dataset" tab. Maybe add one of these blue boxes with something like "If CellCarrier, CellCarrier B or water/PBS were selected in Shape-In, the medium and subsequently, the calculation of Young's modulus cannot be changed."

B-Hartmann, 2022-06-11:
> However, what Shape-Out could at least do is to tell the user that it did not analyze according to the user meta data, but according to the metadata in the .rtdc files.

I think an additional warning message in the "Young's modulus computation"-window is sufficient, what do you think?

paulmueller, 2022-06-12:
Yes, I would agree with that.