DC-analysis/DCscope #131: Windows installation issue

restream93, 2022-07-30:
Hello! I am trying to install the program in my computer and it gives me this error at the end of the installation. I already tried installing the program with admin rights and it did not help. If I ignore the error the program doesn't run after installation. Any clue on how to solve this? 

Thanks

[image]

restream93, 2022-08-01:
I got rid of the error message by using the latest version and also by disabling the antivirus from the computer. However, even when if installation is successful, I cannot run it. It starts but it cannot be run! 

Thanks!

paulmueller, 2022-08-04:
Hi, could you please try to **first** uninstall Shape-Out and **after that** install the latest version again? This could be related to how your Windows is set-up w.r.t security. You might only be able to run Shape-Out when you install it as administrator or when you don't. I'm not sure. What other error message do you get?

paulmueller, 2022-10-19:
Closing due to inactivity

closed 2022-10-19 by paulmueller