DC-analysis/DCscope #132: Problem with dclab `ConfigurationDict` class in ShapeOut

felix-r, 2022-10-19:
I realized that you introduced a new class for the config dictionary (`ConfigurationDict`) in dclab because it broke some of my code 😃 
Apparently it also crashes Shapeout. You can drag datasets into Shapeout but as soon as you want to open them (e.g. via QuickView) you get the following error message below:

```python
Unhandled exception in Shape-Out version 2.11.7:
Traceback (most recent call last):
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\widgets\wait_cursor.py", line 31, in wrapper
    ret = func(*args, **kwargs)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\main.py", line 972, in on_quickview_show_dataset
    self.widget_quick_view.show_rtdc(ds, slot)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\widgets\wait_cursor.py", line 31, in wrapper
    ret = func(*args, **kwargs)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\quick_view\qv_main.py", line 695, in show_rtdc
    self.plot()
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\widgets\wait_cursor.py", line 31, in wrapper
    ret = func(*args, **kwargs)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\quick_view\qv_main.py", line 520, in plot
    self.widget_scatter.plot_data(rtdc_ds=self.rtdc_ds,
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\gui\quick_view\qv_scatter.py", line 95, in plot_data
    x, y, kde, idx = plot_cache.get_scatter_data(
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\plot_cache.py", line 45, in get_scatter_data
    shash = util.hashobj(tohash)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\util.py", line 11, in hashobj
    return hashlib.md5(obj2bytes(obj)).hexdigest()
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\util.py", line 29, in obj2bytes
    return b"".join(obj2bytes(o) for o in obj)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\util.py", line 29, in <genexpr>
    return b"".join(obj2bytes(o) for o in obj)
  File "C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\shapeout2\util.py", line 37, in obj2bytes
    raise ValueError("No rule to convert object '{}' to string.".
ValueError: No rule to convert object '<class 'dclab.rtdc_dataset.config.ConfigurationDict'>' to string.
```

I installed ShapeOut via pip and ran `pip install --upgrade shapeout2` before getting this problem.

felix-r, 2022-10-19:
After downgrading dclab to version 0.46.0 this problem disappears.

paulmueller, 2022-10-19:
Thanks, I have not checked Shape-Out against the latest dclab version.

Which code of yours did dclab 0.46.1 break? It might make sense to add a test for that in dclab to prevent this in the future...

felix-r, 2022-10-19:
My code checked if an object, I get from `rtdc_dataset.config`, is a dictionary:

```python
metadata = rtdc_dataset.config["user"]
if not isinstance(metadata, dict):
    raise TypeError("`metadata` input must be a dictionary,"
                    f"got {type(metadata)} instead.")
```

Since you defined a new class for the config dictionary, this does not work anymore but can easily be fixed.

commit b4ea0a2c (2022-10-19): fix: support latest version of dclab, ironically by enforcing the dependency (close #132)
closed 2022-10-19 by paulmueller