DC-analysis/DCscope #133: Enhancement for Quick View: Allow user to adjust width of plot and event image view

B-Hartmann, 2022-11-28:
Proposed enhancement for Quick View:  
Allow the user to change the width of the plot in order to make more space for the event image. This can be helpful when the user wants to take a closer look at the event image. Currently, the user can zoom in, but the width of the "View" tab is fixed, resulting in a cropped image.

[image]

I think QSplitter or something like that should do the trick and I would like to try to implement that myself.
What do you think?

paulmueller, 2022-11-28:
Thanks for the input. The current Quick-View window is a little hacked-together when it comes to resizing. There is a function somewhere that reduces the size to some hard-coded value when you switch e.g. from plotting to event view. I agree that the current implementation is much less than optimal.

You are very welcome to play around with it and create a PR. You would first have to remove the hard-coded resizing thing. I think this is it:

https://github.com/ZELLMECHANIK-DRESDEN/ShapeOut2/blob/master/shapeout2/gui/quick_view/qv_main.py#L501-L509

Please also feel free to change the arrangement of the GUI elements. If your changes improve user-convenience, I will be happy to include them.

paulmueller, 2022-12-05:
closed by #134

closed 2022-12-05 by paulmueller