DC-analysis/DCscope #137: Last filters not applied when exporting .rtdc files

marketakub, 2023-02-06:
We have found that sometimes the last filters that were applied (box or polygon) were not taken into account when subsequently exporting .rtdc files. No error is shown in ShapeOut2, so the user thinks everything ran correctly, but when the new .rtdc file is examined it is clear that a filter was omitted. Martin has the most experience with this issue

paulmueller, 2023-02-06:
Thanks, can you please point me to a Shape-Out session file (.so2) where this bug occurs?

paulmueller, 2024-03-22:
As mentioned in #153, could it be that the reason for this is that the user exported multiple times to the same directory, overriding existing data there? If yes, then this is fixed in Shape-Out 2.15.4.