DC-analysis/DCscope #139: Problems with reloading inertia ratio

felix-r, 2023-04-12:
dclab 0.48.3 introduced changes in handling the inertia ratio. I currently use v0.48.7. For me, this leads to problems that the features `inert_ratio_raw` and `inert_ratio_cvx` are recalculated even though they are already present in the rtdc-file with the dataset. This is a nuisance because these features will not appear in `ds.features_loaded`, when you don't want to calculate any additional features. But this problem only occurs for some datasets. Here an rtdc file were `inert_ratio_raw` is __not__ loaded:
`guck_division_hsm\\Data\RTDC\20230323_Felix_HL60_LatB_hyper_channels\DMSO\M001_data.rtdc`

Here an example were it is loaded: `guck_division_hsm\\Data\RTDC\20230404_Felix_HL60_LatB_hyper_channels\DMSO\M001_data.rtdc`

Additionally, if the inertia ratios are recalculated, it can fail for cells with big `pos_x`, but not necessarily. In the files above this is not a problem but the problem occurs for this example and all other measurements from this day:
`guck_division_hsm\\Data\RTDC\20230322_Felix_beads_deformation_outside_hyper\S03_set3\inlet\M001_data.rtdc`

[image]

paulmueller, 2023-04-19:
I am not able to reproduce this with Shape-Out 2.13.2 and dclab 0.48.8.
[image]

Could you please answer the following questions:

- Since you are mentioning a specific dclab version, I assume you installed via pip?
- Please paste the output of `python -VV` of your environment.
- Please also paste the output of `pip freeze`.
- Have you tried creating a fresh Python environment?

Note that as far as I understand the issue is two-fold:
1. dclab discards `inert_ratio_raw` and `inert_ratio_cvx` although they were correctly computed by Shape-In.
2. dclab does not correctly compute `inert_ratio_raw` and `inert_ratio_cvx`.

felix-r, 2023-04-19:
Yes, the problem seems two-fold. For point 2.: The values that are computed, the values seem correct. It is just that many values are calculated as NaN. The problem still persists after upgrading to dclab 0.48.8. I didn't try to do this in a fresh environment. Here are the environment infos:
```
> python -VV
> Python 3.9.12 (main, Apr  4 2022, 05:22:27) [MSC v.1916 64 bit (AMD64)]
``` 

```
> pip freeze
absl-py==1.4.0
appdirs==1.4.4
argcomplete==2.0.0
astunparse==1.6.3
cachetools==5.3.0
certifi @ file:///C:/Windows/TEMP/abs_e9b7158a-aa56-4a5b-87b6-c00d295b01fanefpc8_o/croots/recipe/certifi_1655968940823/work/certifi
cffi==1.15.1
charset-normalizer==2.1.0
click==8.1.3
colorama==0.4.5
dclab==0.48.8
distlib==0.3.5
fcswrite==0.6.1
filelock==3.7.1
flatbuffers==23.1.21
gast==0.4.0
google-auth==2.16.0
google-auth-oauthlib==0.4.6
google-pasta==0.2.0
grpcio==1.51.1
h5py==3.7.0
hdf5plugin==3.3.1
idna==3.3
imageio==2.4.1
importlib-metadata==6.0.0
Jinja2==3.1.2
keras==2.11.0
libclang==15.0.6.1
Markdown==3.4.1
MarkupSafe==2.1.1
npTDMS==1.6.1
numpy==1.23.1
oauthlib==3.2.2
opt-einsum==3.3.0
packaging==21.3
pathlib==1.0.1
Pillow==9.4.0
platformdirs==2.5.2
protobuf==3.19.6
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
pyparsing==3.0.9
PypeR==1.1.2
PyQt5==5.15.7
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.0
pyqtgraph==0.12.4
pytz==2022.1
pytz-deprecation-shim==0.1.0.post0
requests==2.28.1
requests-oauthlib==1.3.1
rpy2==3.5.2
rsa==4.9
scipy==1.8.1
shapeout2==2.13.2
six==1.16.0
tensorboard==2.11.2
tensorboard-data-server==0.6.1
tensorboard-plugin-wit==1.8.1
tensorflow==2.11.0
tensorflow-estimator==2.11.0
tensorflow-intel==2.11.0
tensorflow-io-gcs-filesystem==0.30.0
termcolor==2.2.0
typing_extensions==4.4.0
tzdata==2022.1
tzlocal==4.2
urllib3==1.26.10
userpath==1.8.0
virtualenv==20.15.1
virtualenv-clone==0.5.7
Werkzeug==2.2.2
wincertstore==0.2
wrapt==1.14.1
zipp==3.11.0
```

paulmueller, 2023-04-19:
Thanks for the input. That does not make sense. Could you please run the following Python script and paste back the output here?

```python
import time

import dclab
import numpy as np

path = "guck_division_hsm\\Data\RTDC\20230322_Felix_beads_deformation_outside_hyper\S03_set3\inlet\M001_data.rtdc"
path = "M001_data.rtdc"

print("dclab", dclab.__version__)

ds = dclab.new_dataset(path)

print("sample:", ds.config["experiment"]["sample"])
print("date:", ds.config["experiment"]["date"])
print("time:", ds.config["experiment"]["time"])
print("md5-5m:", dclab.util.hashfile(path, count=80))

print("inert_ratio_raw available:", "inert_ratio_raw" in ds)
print("inert_ratio_raw loaded:", "inert_ratio_raw" in ds.features_loaded)

t1 = time.time()
print("Computing inertia ratio, please wait..")
inert_ratio_raw = ds["inert_ratio_raw"]
print(f"Computation took {time.time()-t1:.1f} seconds.")

print("Min", np.nanmin(inert_ratio_raw))
print("Max", np.nanmax(inert_ratio_raw))
print("Mean", np.nanmean(inert_ratio_raw))
print("NaNs", np.sum(np.isnan(inert_ratio_raw)))
```

felix-r, 2023-04-20:
I ran the code in the same environment. Here's the output:

```
dclab 0.48.8
sample: inlet
date: 2023-03-22
time: 11:44:54
md5-5m: 4f7f15ad6962206a44e9c7e583e77ff5
inert_ratio_raw available: True
inert_ratio_raw loaded: False
Computing inertia ratio, please wait..
C:\Users\freiche\Anaconda3\envs\so2\lib\site-packages\dclab\features\inert_ratio.py:316: RuntimeWarning: invalid value encountered in sqrt
  inert_ratio_raw[ii] = np.sqrt(moments["mu20"]/moments["mu02"])
Computation took 35.4 seconds.
Min 0.6971645948218216
Max 1.3141488410210778
Mean 0.9996123666323039
NaNs 22826
```

I also created a fresh environment with python 3.10 and get the same output.

paulmueller, 2023-04-20:
Thanks, apparently Windows uses 32bit ints as default when creating new arrays. I fixed this in dclab 0.49.0 and created two additional issues in the dclab repository.

felix-r, 2023-04-20:
Wow, dealing with this floating precision is a real pain in the ass!

paulmueller, 2023-04-26:
With the latest version of dclab, the first part of the problem should also be solved.

closed 2023-04-26 by paulmueller