DC-analysis/DCscope #145: Failure at startup

matteounimib, 2023-10-19:
>> pipx install shapeout2
>> .local/bin/shapeout2
It open a windows with the following error.

`Unhandled exception in Shape-Out version 2.13.6:
Traceback (most recent call last):
  File "/home/darkflash/.local/bin/shapeout2", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/me/.local/pipx/venvs/shapeout2/lib/python3.11/site-packages/shapeout2/__main__.py", line 35, in main
    window = ShapeOut2(*app.arguments()[1:])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/me/.local/pipx/venvs/shapeout2/lib/python3.11/site-packages/shapeout2/gui/main.py", line 88, in __init__
    dclab.rtdc_dataset.fmt_dcor.APIHandler.add_api_key(api_key)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'dclab.rtdc_dataset.fmt_dcor' has no attribute 'APIHandler'
`

paulmueller, 2023-10-20:
Hi, this is due to a regression dclab. An intermediate solution is to downgrade dclab to a version below 0.53.0

`pipx install shapeout2 "dclab<0.53.0"`

paulmueller, 2023-10-20:
This bug is already fixed in the `master` branch.

paulmueller, 2023-10-24:
This is fixed in Shape-Out 2.13.7

closed 2023-10-24 by paulmueller