DC-analysis/DCscope #154: Statistics under 'Quick View' in Shape-Out 2.15.0 doesn't update

rgoswam1, 2024-01-11:
By default, the measurements show Area vs Deformation plot and the related statistics under the 'Quick View' tab. However, when loading a new measurement as shown in image **Capture_01** (Measurement no. 4 (0.16 µl/s flow rate)) and potting the Area vs Young's modulus [kPa], the plot is updated but the statistics are not.
[image]

**Capture_02** shows that the parameters under 'Plot' were selected as Area vs. Young's modulus [kPa], and then the measurement (no. 5 (0.2 µl/s)) was loaded. In this case, the statistics are updated.
[image]

**Capture_03** shows, for the same measurement (no. 5 (0.2 µl/s)), if the plotting parameters are changed from Area vs Young's modulus [kPa] to Area vs Deformation, the plot is updated but the statistics are not.
[image]

paulmueller, 2024-01-11:
Thanks, I missed this when introducing caching for the statistics in 2.15.0.

paulmueller, 2024-01-12:
The features used for plotting were not part of the cache key. This will be fixed in 0.15.1.

commit 2ae58288 (2024-01-12): fix: statistics not updating when changing plot axes (close #154)
closed 2024-01-12 by paulmueller