DC-analysis/DCscope #157: Shape-Out adds prefix "so2exp" to every log entry when exporting data, but does not create dedicated log entry

B-Hartmann, 2024-02-08:
Shape-Out 2.15.1
Windows 10

I filtered and exported a file and noticed in the logs (using DCKit 0.16.0) that a string "so2exp" is added to the title of all existing log entries. 
I do not think this is intended and it does not make sense to me, I would expect that Shape-Out creates a log entry for the export.
And if the exported file is exported by Shape-Out again, it will add this prefix again 

[image]
After second export:
[image]

B-Hartmann, 2024-02-14:
Yes, a separate entry would be more consistent with the style of logs in RTDC files (every action that changes the file creates a new entry in the logs).

I checked how the logs look like for the case of joining two datasets after exporting:
[image]

I like how the logs from the source files have the prefix "src-#1" and so on, it allows users to quickly identify which entry belongs to which source file. I recommend that the entries "cfg_src-#1" are named in the same fashion, e.g. "src-#1_cfg" to be consistent and to have all logs related to the same source file together in the list of logs.

Would it make sense to have an entry called "file history", in which all other entries are listed together with a time stamp so that one can see what has happened to the file in chronological order, especially in files which were processed many times? Theoretically, this information is in the file, but spread out over all log entries and the metadata section of the file (entries "shapein-acquisition" and "SoftwareSettings" do not contain a time stamp, but the creation time of the file is in the metadata section). (Actually, after joining, the time and date of acquisition can be found in "cfg_src-#1".)

I mean something like
```
2024-02-08 14:12:13 - src-#1_shapein_acquisition
2024-02-08 14:12:13 - src-#1_M005_SoftwareSettings.ini
2024-02-08 16:15:13 - src-#2_shapein_acquisition
2024-02-08 16:15:13 - src-#2_M006_SoftwareSettings.ini
2024-02-14 07:02:21 - dclab-join
```

paulmueller, 2024-03-22:
I have created https://github.com/DC-analysis/dclab/issues/251 to reflect your point on having a separate log entry for every export.

paulmueller, 2024-03-22:
I have created https://github.com/DC-analysis/dclab/issues/252 for your request to have something like a time trace on the logs.

commit 9155e082 (2024-03-22): fix: remove the so2exp prefix when exporting data (close #157)
closed 2024-03-22 by paulmueller