DC-analysis/DCscope #158: Exception when loading dataset with flow rate zero/infinity

B-Hartmann, 2024-03-15:
Shape-out 2.15.3
Windows 10

Loading a dataset with flow rate zero (syringe pumps were not used) sometimes causes the following exception:

```python
Unhandled exception in Shape-Out version 2.15.3:
Traceback (most recent call last):
  File "shapeout2\gui\widgets\wait_cursor.py", line 31, in wrapper
    ret = func(*args, **kwargs)
  File "shapeout2\gui\main.py", line 309, in adopt_pipeline
    self.widget_ana_view.adopt_pipeline(pipeline_state)
  File "shapeout2\gui\analysis\ana_view.py", line 30, in adopt_pipeline
    self.update_content()
  File "shapeout2\gui\analysis\ana_view.py", line 40, in update_content
    self.widget_meta.update_content()
  File "shapeout2\gui\analysis\ana_meta.py", line 105, in update_content
    self.update_info_box(self.groupBox_setup, cfg,
  File "shapeout2\gui\analysis\ana_meta.py", line 61, in update_info_box
    k, v, t = format_config_key_value(section, key, value)
  File "shapeout2\gui\analysis\ana_meta.py", line 123, in format_config_key_value
    dec = int(np.ceil(np.log10(1/np.abs(value))))
OverflowError: cannot convert float infinity to integer
```

I can just click "Close" and the dataset is loaded, so I can still work with the data. Since I always use a pressure pump for my measurements and I always check those files with Shape-Out, it is just very inconvenient to dismiss the error message all the time.

In this context, I noticed that actually not all of my measurements cause this issue. If the values in the hdf5 file are 0.0 for all three flow rate values (total flow, sample and sheath), then there is no problem. But if sample or sheath flow rate are something like `4.9E-323` (which I guess is Pythons way to say -Inf), then this problem arises.

B-Hartmann, 2024-03-15:
The best way to solve this issue would be if ShapeIn would save well defined values when creating the file in the first place, of course.

phidahl, 2024-03-15:
4.9E-323 is not -Inf but very very close to zero. I don't know what ShapeOut is doing with these values, but there surely a wait to handle valid floating point values gracefully.
Floating point numbers (especially when they are a result of a calculation) should not be directly compared due to their inaccuracy. See also https://en.wikipedia.org/wiki/Floating-point_arithmetic#Accuracy_problems If one would like to check if a floating point number is zero it would be best to assume an interval.

B-Hartmann, 2024-03-15:
As a user, I would welcome it if both ShapeIn and Shape-Out would consider values below 1E-10 (µl/s) as 0. It is not possible to operate the device at these low flow rate. So, if this value would be saved by ShapeIn when the pumps are not moving and if Shape-Out would handle these values in the existing files, that would be awesome.

I am aware this is very low in priority for you guys, but in terms of user experience, it makes a difference if you have to dismiss 15 error messages when loading 15 datasets 😃

commit e1aa31e2 (2024-03-20): fix: very small numbers not parsed when showing metadata (close #158)
closed 2024-03-20 by paulmueller