DC-analysis/DCscope #165: ref: migrate from PyQt5 to PyQt6

paulmueller, 2024-09-19:
The latest version of QtCreator does not support PyQt5 (the resulting .ui files cannot be used due to additional type definitions).
Therefore, it would be good to migrate the codebase and all .ui files to PyQt6.
For the differences between the two and some technical details for the migration, see e.g. https://www.pythonguis.com/faq/pyqt5-vs-pyqt6/

RaghavaAlajangi, 2024-10-23:
Hi @paulmueller 
So, basically, it is all about replacing components mentioned in the link you provided, right? Or do we need to run the app with PyQt6 and check where it is throwing errors?

paulmueller, 2024-10-23:
Yes. To be clear, it means
1. opening all .ui files with the latest version of QtCreator and saving it back (Qt6 automatically adds type hints and maybe other things)
2. in pyqt6, several imports and attributes are different and need to be imported from other locations in pyqt6 compared to where they were before in pyqt5 (the link above is only an example, there are probably better resources for migrating pyqt5 to pyqt6)

paulmueller, 2024-11-04:
Current road block: https://forum.qt.io/topic/158135/new-update-of-qt-creator-causes-problem-with-pyqt6-uic/4

paulmueller, 2024-11-04:
Running `python -m shapeout2 --version` still causes a segmentation fault.

paulmueller, 2024-11-04:
Everything is currently sitting in the `pyqt6` branch.

paulmueller, 2024-11-05:
Segmentation fault is fixed with ad8eaf5

paulmueller, 2024-11-05:
I managed to make everything work, but we are currently restricted to using QtCreator 8.

paulmueller, 2024-11-05:
done in #177

closed 2024-11-05 by paulmueller