DC-analysis/DCscope #182: Basins: Questions and Suggestions

B-Hartmann, 2024-12-06:
Shape-Out 2.19.1

What features are _Basin mapping x_?
[image]

Would it make sense to write in the tab _Meta_ in Analysis View if a file contains basins so that users have a way of checking if the file is just small or contains basins and maybe where the remote file should be so that the basins work?

paulmueller, 2024-12-06:
These features define which indices in the basin correspond to the current index: https://dclab.readthedocs.io/en/stable/sec_av_basins/index.html#definitions

paulmueller, 2024-12-06:
IMO adding a tab in analysis view that lists all basins makes a lot of sense.

commit c98ea15e (2025-01-06): feat: implement showing dataset basins in the analysis view (close #182)
closed 2025-01-06 by paulmueller