DC-analysis/DCscope #207: Plotting: Contour line computation can crash Shape-Out

paulmueller, 2025-02-27:
Create a plot of index vs index. When contour computation is active, Shape-Out becomes unresponsive. The problem is that for such a narrow (1D) signal, no sane contour can be computed.

This probably boils down to having a safeguard implemented in dclab (either check for feature identity or set an iteration limit for the algorithm or allow to abort it).

paulmueller, 2025-03-13:
This will not be enough, because the user could also select "time vs index". There should be a smart solution to this.

paulmueller, 2025-03-16:
I would say all of the three ideas are important.

Identity: To catch the trivial cases, the contour computation should not be done when both features are the same.

Iteration Limit: The next step is to set a default iteration limit, which (if reached) Shape-Out will have to handle gracefully.

Aborting: Allowing to abort the computation in the Shape-Out GUI would be a nice-to-have feature. But in order for that to work, the computation must be moved in a separate thread.

RaghavaAlajangi, 2025-03-17:
Then, does this issue have to be divided into 3 PRs (dclab & shapeout2) or all of them in one?

paulmueller, 2025-03-17:
Yes, a separation into multiple PRs makes sense. The functionality has to be made available in dclab first, so this is definitely be a multi-PR issue.